ISU-MAVRIC/Old-MAVRIC-Systems #3: Power budget design

Talonj123, 2017-08-07:
@matgyver can you take a look at the power budget spreadsheet?
https://github.com/m2i/MAVRIC-Electrical/blob/master/Electrical/PowerBudget.xlsm

I did everything in terms of power instead of current to allow for regulators and such. I know that for switches, the voltage is irrelevant, but the math can be done at the voltage of the switched line. It seemed like a simple way to make it all work.

Talonj123, 2017-08-07:
If you have a preferred template, I'd love to use it.

matgyver, 2017-08-07:
I would keep track of both voltage and current as well.  In a large system
like this, you will have multiple regulators and different voltages.  You
will want to track the current draw on each rail.  I've added you to a
HABET one we did a few years ago.

Talonj123, 2017-08-07:
Is that here on GitHub?

matgyver, 2017-08-07:
Google Spreadsheet, check your gmail.

Talonj123, 2017-08-08:
I've updated it to call out current and voltage (see the Elaboration tab). I'll post a detailed guide on the wiki

closed 2017-08-08 by Talonj123